rusefi/hw_modular #4: unify capacitors if possible

rusefi, 2019-05-27:
at the moment BOM seems to have three groups of same uF but different voltages. Could this be unified?

9;C16,C14;C_0603_1608Metric_Pad1.05x0.95mm_HandSolder;2;100nF 100V;
11;C13,C10,C6,C5,C4,C1;C_0603_1608Metric_Pad1.05x0.95mm_HandSolder;6;100nF 50V;

15;C3;C_0603_1608Metric_Pad1.05x0.95mm_HandSolder;1;4.7nF;
12;C11;C_0603_1608Metric_Pad1.05x0.95mm_HandSolder;1;4.7nF 50V;

13;C9,C8;C_0603_1608Metric_Pad1.05x0.95mm_HandSolder;2;10uF 25V ;
14;C7;C_0603_1608Metric_Pad1.05x0.95mm_HandSolder;1;10uF 50V ;

rusefi, 2019-05-27:
@960

960, 2019-05-27:
Yes, this is a fast and easy fix.

It's just because of different naming, uppercase/lovercase etc.

I can fork and update